shogunxam/ESP32_BLETracker #5: BLE Tracker

DivanX10, 2020-12-03:
Tell me, with your firmware, any BLE tracker will work or certain models of BLE trackers? Какой BLE трекер порекомендуете, если заказать на aliexpress?

shogunxam, 2020-12-04:
Not all the BLE devices are compatible, the scan check for the MAC Address of the device, and some devices have a changing BLE address (normally called random instead of public address) for privacy/security reason. The same for the reading of the battery level, the device has to expose the Battery Service (0x180F) and the Battery Level characteristic (0x2A19),
I have tested successfully with:
- nut mini (work both discovery and battery level).
- Amazfit bit (works only the discovery)
- Xiomi Mi Smart Band 4 (works only the discovery)

DivanX10, 2020-12-05:
I bought a Redmond SkyTracker 08S tracker. Works. Displays the on and off status, and displays the RSSI signal strength. The battery status is always -1. Do I understand correctly that the battery status is not displayed for all devices? Although the app itself displays the battery level of the REDMOND RFT-08S tracker.

shogunxam, 2020-12-05:
Yes, the battery level can be read only from the devices providing the Battery Service (0x180F) and the Battery Level characteristic (0x2A19), check the availability using a nRF Sniffer i.e. [nRF Connect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp). Also in this case can happen the esp32 fails to pair with the device so the battery cannot be read. Unfortunately I haven't a lot of devices to test.
I will add your device to the list of working devices :smiley:

DivanX10, 2020-12-05:
Installed nRF Connect and connected to the tracker. I didn't find a line about the battery. There are lines Generic Access, Device Name, Appearance, Peripheral, Preferred Connection, Generic Attribute, Nordic UART Service, TX Characteristics, RX Characteristics.

shogunxam, 2020-12-05:
Within nRF Connect you have to connect to the discovered device (it can take a while) Then you have to search for the highlighted items.
[image]

DivanX10, 2020-12-05:
Thanks for the screenshots, now everything is clear. Well, this is not critical, the main thing is that in your firmware I see the tracker and there is an on\off status. Added a mac to the white list and extra Macs are eliminated. Very convenient. For example, there is a firmware OpenMQTTGateway, it is useless, there is no normal connection with the tracker and a lot of extra. Thank you for the excellent firmware.

DivanX10, 2020-12-27:
I connected the itag tracker. Only the status works, the battery level is not displayed, although through nRF Connect there is a display of the battery level. Will it be possible to display the battery level of the itag tracker?

[image]

These are the topics I have displayed in the MQTT client
[image]

shogunxam, 2021-01-02:
Perhaps the ESP is failing to connect to the itag unfortunately I haven't one to test. I have seen a similar issue with the nut (non nut mini) of another guy but I haven't found a solution yet.

shogunxam, 2021-01-02:
I forgot, obviously you have added the itag's MAC address to the list of the devices for which the tracker has to read the batteries, right?

DivanX10, 2021-01-18:
Вы не указали еще BLE трекер iTag. Данный трекер тоже обнаруживается в вашей прошивке.
You haven't specified the iTag BLE tracker yet. This tracker is also found in your firmware.
[image]

shogunxam, 2021-01-18:
Is the itag only discovered or is the battery value read also?

DivanX10, 2021-01-18:
Only iTag is detected, and the battery is not read. The value -1 is displayed.
Обнаруживается только iTag, а батарея не считывается. Отображается значение -1.
[image]

shogunxam, 2021-03-03:
This issue has been marked as stale because it has not had
  recent activity. It will be closed if no further activity occurs. Thank you
  for your contributions.

closed 2021-03-03 by shogunxam