shogunxam/ESP32_BLETracker #6: Settings via the web interface

DivanX10, 2020-12-08:
is it possible to modify the firmware so that the web interface can specify the IP and port mqtt, specify the white and black list, specify the mac address? It's just inconvenient to flash esp32 every time to enter the mac address in the white list and specify a different address of the mqtt server

shogunxam, 2020-12-09:
Yes this is a good idea for further improvements. 
Bay the way the scan interval can be changed changing the macro 

```#define BLE_SCANNING_PERIOD   10```

in the config.h

DivanX10, 2020-12-09:
Yes, I know that. It was just easier to configure via the web interface, and then every time you fill in a sketch because of the settings, it's not very convenient. Let's say I already set up everything, but then I bought another BLE tracker and to add it to the white list, I will have to fill in the sketch again with code editing.

DivanX10, 2020-12-09:
I really liked your firmware. Before I found your firmware, I tried the firmware on github of different authors and everything is wrong. It would be great if you improve your firmware. For example, add the function of scanning phones (if possible), then there will be more opportunities, because someone will use trackers, someone phones. I have my own [YouTube channel about smart home](https://www.youtube.com/channel/UCPoTSYr_VucFPbrpQ30aE9Q) and I'm going to tell you about your firmware on my channel. Bluetooth tags, this is an important function for determining the home friend or foe. A smart home can identify its owner by scanning a phone or tracker, but almost everyone has phones, and not everyone has trackers + they are not reliable, often break down or the battery runs out at the wrong time.

shogunxam, 2020-12-12:
I will see what I can do during the Xmas holidays :christmas_tree:  :smiley:

shogunxam, 2020-12-21:
I hope this can satisfy your request
https://github.com/shogunxam/ESP32_BLETracker/pull/7

closed 2020-12-21 by shogunxam

DivanX10, 2020-12-21:
Thanks). I tried adding in home assistant and everything works fine. I did not write everything in the configuration file configuration.yaml, and everything was taken out separately. Created the includes folder and bin_sensor and sensors subfolders. Inside them, I created two ble_tracker files.yaml where I added the code and also created a separate mqtt file for MQTT

**It looks like this for me**
[image]

**In the configuration file configuration.yaml inserted these lines**
binary_sensor: !include_dir_merge_list includes/bin_sensor
sensor: !include_dir_merge_list includes/sensors/

[image]
[image]
[image]

**In the folder HomeAssistant\includes\bin_sensor\ble_tracker.yaml**
#####BLE Tracker#####
#Binary Sensors to resolve unknown device state
#######################
#Check BLETracker state
- platform: mqtt
  name: BLETracker_state
  state_topic: 'home/BLETracker/LWT'
  payload_on: "online"
  payload_off: "offline"
  device_class: "connectivity"

**In the folder HomeAssistant\includes\sensors\ble_tracker.yaml**
#####BLE Tracker#####
#Sensors
- platform: mqtt
  state_topic: 'home/BLETracker/mac'
  name: 'BLETracker_Redmond_state' 
  value_template: "{{ value_json.state }}"
  expire_after: 120 
  force_update: true

- platform: mqtt
  state_topic: 'home/BLETracker/mac'
  name: 'BLETracker_Redmond_battery' 
  value_template: "{{ value_json.battery }}"
  force_update: true

**In the folder HomeAssistant\includes\sensors\mqtt.yaml**
#####BLE Tracker#####
 - platform: mqtt
   name: "BLETracker_Redmond"
   state_topic: "home/BLETracker/mac/state"
   availability_topic: "home/BLETracker/LWT"
   expire_after: 300
   value_template: >-
      {% if value == 'on' %}
         home
      {% elif value == 'off' %}
         away
      {% else %}
         away
      {% endif %}