shogunxam/ESP32_BLETracker #8: The status of the gateway is unstable

DivanX10, 2020-12-26:
I noticed that the gateway status changes frequently, almost every 30-40 seconds. You can somehow fix it so that it is stable. I tried changing the time in the config, but it didn't help. How do I fix this?

here I specified 5 minutes
#define WIFI_CONNECTION_TIME_OUT  300000 /*30 seconds*/

#Check BLETracker state
- platform: mqtt
  name: BLETracker_state
  state_topic: 'home/BLETracker/LWT'
  payload_on: "online"
  payload_off: "offline"
  device_class: "connectivity"

[image]

shogunxam, 2021-01-02:
Which version are you using? The one on the master or develop branch?
WIFI_CONNECTION_TIME_OUT  is just a timeout used to reset the ESP32 in case it's no able to connect to the wi-fi. 
In case you have build with Arduino Ide be sure to use my modified BLE library as described in the documentation.
To investigate the issue could be useful take a look the logs in the console. The disconnection could be caused by a crash in this case if you he build with PlatformIO firmware.elf and the back trace printed on the console can be used to find where is the issue.
Anyway I have never encountered so many disconnection. 
This is the state of my tracker using the latest develop version. Disconnections are forced by me after an OTA update.
[image]

SmartM-ui, 2021-01-02:
Hi, 
[image]
I have checked and my ESP is stable since its start

DivanX10, 2021-01-18:
Thank you for your hard work that added the ability to add a mac via the web interface. I have uploaded the latest version of your firmware version 2.1 via the Arduino IDE. I still have frequent breaks. The library I used [this one you specified] (https://github.com/shogunxam/ESP32-BLE-Arduino)

Here I put the esp32_blu_arduino library
[image]

I filled in the latest firmware version 2.1, in the settings I specified only the SSID and password and the address of the MQTT server
[image]

---------------

Спасибо за ваш труд, что добавили возможность добавления mac через веб интерфейс. Я залил последнюю версию вашей прошивки версии 2.1 через Arduino IDE. У меня по прежнему частые разрывы. Библиотеку я использовал [эту, которую вы указали](https://github.com/shogunxam/ESP32-BLE-Arduino)

Вот сюда я положил библиотеку ESP32_BLE_Arduino
[image]

Залил последнюю прошивку версии v2.1, в настройках я указал только SSID и пароль и адрес MQTT сервера
[image]

shogunxam, 2021-01-18:
The last time I had a similar problem was router's issue.
Could you please check the logs? You have to enable the debug level for logs from the UI. So we can understand if it's a crash or only a disconnection from the MQTT broker.
I suggest also to build with Platform.IO because it uses newer libraries. The BLE Library is the correct one.
Is the Wi-Fi signal strength enough between the ESP and the access point? Mine is 2 meters from the router and I haven't a disconnection since 2 days

DivanX10, 2021-01-18:
Вот что пишется в логах. MQTT брокер у меня установлен на synology в докере и имеет порт 2883. Другие шлюзы как WLED, Redmond for Sky работают нормально

Here is what is written in the logs. The MQTT broker I have installed on the synology in the Docker and has a port 2883. Other gateways like WLED, Redmond for Sky work fine

Timestamp | Message
-- | --
2021-01-18 13:59:25 | New LogLevel configuration succesfully saved.
2021-01-18 13:59:12 | Device 04CF8C20939E within range, RSSI: -88
2021-01-18 13:59:01 | Devices 04CF8C20939E is gone out of range
2021-01-18 13:59:01 | MQTT broker connected!
2021-01-18 13:59:01 | Error: MQTT broker disconnected, connecting...
2021-01-18 13:57:54 | MQTT broker connected!
2021-01-18 13:57:54 | Error: MQTT broker disconnected, connecting...
2021-01-18 13:56:47 | MQTT broker connected!
2021-01-18 13:56:47 | Error: MQTT broker disconnected, connecting...
2021-01-18 13:55:41 | Devices 0859598B97D1 is gone out of range
2021-01-18 13:55:41 | MQTT broker connected!
2021-01-18 13:55:41 | Error: MQTT broker disconnected, connecting...
2021-01-18 13:54:34 | Devices 0C50C4297413 is gone out of range
2021-01-18 13:54:34 | Devices 7EFD5F3F5926 is gone out of range
2021-01-18 13:54:34 | MQTT broker connected!
2021-01-18 13:54:34 | Error: MQTT broker disconnected, connecting...
2021-01-18 13:53:56 | Discovered new device 04AE128AD5B5 within range, RSSI: -79
2021-01-18 13:53:49 | Discovered new device 609D8261679E within range, RSSI: -91
2021-01-18 13:53:27 | MQTT broker connected!
2021-01-18 13:53:27 | Error: MQTT broker disconnected, connecting...
2021-01-18 13:52:21 | MQTT broker connected!
2021-01-18 13:52:21 | Error: MQTT broker disconnected, connecting...
2021-01-18 13:51:34 | Device 0C50C4297413 within range, RSSI: -88
2021-01-18 13:51:14 | Devices 0C50C4297413 is gone out of range
2021-01-18 13:51:14 | MQTT broker connected!
2021-01-18 13:51:13 | Error: MQTT broker disconnected, connecting...
2021-01-18 13:50:06 | MQTT broker connected!
2021-01-18 13:50:06 | Error: MQTT broker disconnected, connecting...
2021-01-18 13:49:00 | MQTT broker connected!
2021-01-18 13:48:58 | Error: MQTT broker disconnected, connecting...
2021-01-18 13:48:52 | Discovered new device 0C50C4297413 within range, RSSI: -93
2021-01-18 13:47:52 | MQTT broker connected!
2021-01-18 13:47:52 | Error: MQTT broker disconnected, connecting...
2021-01-18 13:46:45 | MQTT broker connected!
2021-01-18 13:46:45 | Error: MQTT broker disconnected, connecting...
2021-01-18 13:45:39 | MQTT broker connected!
2021-01-18 13:45:39 | Error: MQTT broker disconnected, connecting...
2021-01-18 13:44:33 | Devices 3264431EBB8D is gone out of range
2021-01-18 13:44:32 | MQTT broker connected!
2021-01-18 13:44:32 | Error: MQTT broker disconnected, connecting...
2021-01-18 13:43:38 | Discovered new device 0859598B97D1 within range, RSSI: -79
2021-01-18 13:43:26 | MQTT broker connected!
2021-01-18 13:43:26 | Error: MQTT broker disconnected, connecting...
2021-01-18 13:42:20 | MQTT broker connected!
2021-01-18 13:42:20 | Error: MQTT broker disconnected, connecting...
2021-01-18 13:41:13 | Devices 500B419D9688 is gone out of range
2021-01-18 13:41:13 | MQTT broker connected!
2021-01-18 13:41:13 | Error: MQTT broker disconnected, connecting...
2021-01-18 13:40:07 | MQTT broker connected!
2021-01-18 13:40:07 | Error: MQTT broker disconnected, connecting...
2021-01-18 13:39:10 | Discovered new device 7EFD5F3F5926 within range, RSSI: -87
2021-01-18 13:39:00 | MQTT broker connected!
2021-01-18 13:39:00 | Error: MQTT broker disconnected, connecting...
2021-01-18 13:37:54 | Devices F82441C5320D is gone out of range
2021-01-18 13:37:54 | MQTT broker connected!
2021-01-18 13:37:54 | Error: MQTT broker disconnected, connecting...
2021-01-18 13:36:48 | MQTT broker connected!
2021-01-18 13:36:48 | Error: MQTT broker disconnected, connecting...
2021-01-18 13:35:41 | MQTT broker connected!
2021-01-18 13:35:41 | Error: MQTT broker disconnected, connecting...
2021-01-18 13:35:22 | Discovered new device F82441C5320D within range, RSSI: -72
2021-01-18 13:34:35 | MQTT broker connected!
2021-01-18 13:34:35 | Error: MQTT broker disconnected, connecting...
2021-01-18 13:34:30 | Device FFFFC0014EAD within range, RSSI: -74
[... 15 lines cut ...]
2021-01-18 13:25:46 | MQTT broker connected!
2021-01-18 13:25:45 | Error: MQTT broker disconnected, connecting...
2021-01-18 13:24:39 | MQTT broker connected!
2021-01-18 13:24:39 | Error: MQTT broker disconnected, connecting...
2021-01-18 13:23:32 | MQTT broker connected!
2021-01-18 13:23:32 | Error: MQTT broker disconnected, connecting...
2021-01-18 13:22:26 | MQTT broker connected!
2021-01-18 13:22:26 | Error: MQTT broker disconnected, connecting...
2021-01-18 13:21:20 | MQTT broker connected!
2021-01-18 13:21:20 | Error: MQTT broker disconnected, connecting...
2021-01-18 13:20:14 | Devices 4F07DF0F43FA is gone out of range
2021-01-18 13:20:14 | MQTT broker connected!
2021-01-18 13:20:14 | Error: MQTT broker disconnected, connecting...
2021-01-18 13:20:10 | Device 04CF8C20939E within range, RSSI: -86
2021-01-18 13:19:08 | MQTT broker connected!
2021-01-18 13:19:08 | Error: MQTT broker disconnected, connecting...
2021-01-18 13:18:44 | Discovered new device 500B419D9688 within range, RSSI: -89
2021-01-18 13:18:02 | MQTT broker connected!
2021-01-18 13:18:02 | Error: MQTT broker disconnected, connecting...
2021-01-18 13:16:56 | MQTT broker connected!
2021-01-18 13:16:56 | Error: MQTT broker disconnected, connecting...
2021-01-18 13:15:50 | MQTT broker connected!
2021-01-18 13:15:50 | Error: MQTT broker disconnected, connecting...
2021-01-18 13:14:44 | MQTT broker connected!
2021-01-18 13:14:44 | Error: MQTT broker disconnected, connecting...

shogunxam, 2021-01-18:
I don't see log about restart caused by a crash so it's only a problem of connection. I see you haven't set MQTT_USERNAME and MQTT_PASSWORD, haven't you set credential in your broker?

DivanX10, 2021-01-18:
I use MQTT without authorization.

shogunxam, 2021-01-18:
Could you please play with the value of MQTT_KEEPALIVE and MQTT_SOCKET_TIMEOUT you find at the top of the file 
ESP32_BLETracker/main/main.ino

Below there is an explanation of the two parameters.
If you are using Arduino IDE I suggest to upgrade the PubSubClient library to the latest version (currently in Platform.IO configuration I'm using  the version 2.8).
Moreover in always in the in Platform.IO configuration I'm using MQTT_KEEPALIVE=30 and MQTT_MAX_PACKET_SIZE=1024, you can add the last one in the main.ino under the others simply adding the line
#define MQTT_MAX_PACKET_SIZE 1024

MQTT_SOCKET_TIMEOUT is how long the client will wait for data to arrive. For example, after sending the initial CONNECT, it is how long it will wait for the CONACK to arrive before deciding the connection is dead. Or when trying to read a complete packet from the network, how long it will wait between individual bytes. It isn't related to anything regarding the MQTT protocol itself - its just basic TCP timeout handling solely on the client.

MQTT_KEEPALIVE is an MQTT protocol defined timeout. It is how long either end of the connection should wait without having received any data from the other side before deciding the connection is dead. So to be clear - this is when the underlying TCP connection is still open, but no data is flowing. When the keepalive timer expires (typically, a 2x grace period is used), the connection is considered dead and should be closed. As an 'abnormal' disconnection, it will trigger any Will message provided by the client on the original connect. The client includes its keepalive timer value as part of the connect packet - so the broker knows what keepalive interval to expect the client to honour.
Just to reiterate, this along applies whilst the TCP connection is open - if the TCP connection drops, they the Will will send straight away (assuming the client didn't first send a proper MQTT Disconnect packet).

shogunxam, 2021-01-18:
Taking a look to this post 
https://github.com/maximelouet/iot-sitting-posture-assistant/commit/0f7714c0b487c3c92604bb0e283cc61f2f1d170e
the problem was solved lowering the MQTT_KEEPALIVE  to 10

DivanX10, 2021-01-18:
Roger that. I will try all the options and unsubscribe to you.

SmartM-ui, 2021-01-18:
Hi,
I checked the log of my two ESP32s with the latest version of your great release and have no disconnections on both ESPs.
They work perfectly.
Thanks again for your hard work.

DivanX10, 2021-01-18:
Ура!! Заработало как надо. 
Hurray!! It worked as it should. 🥳

В Arduino IDE изменил параметр MQTT_KEEPALIVE to 10, как вы и написали и все заработало. Теперь нет частых разрывов.
In the Arduino IDE, I changed the parameter MTU_KEEP ALIVE to 10, as you wrote, and everything worked. Now there are no frequent breaks.
[image]

Из истории видно, что разрывов нет
The history shows that there are no gaps
[image]

shogunxam, 2021-01-18:
Glad to hear that. I will do some test with this value too, and if it continues to work as it should I will set 10 as default.
I close thi issue.

closed 2021-01-18 by shogunxam