shogunxam/ESP32_BLETracker #32: Friendly name to the MAC address

SmartM-ui, 2021-03-18:
Hi,
would it be possible to associate a friendly name for each NUT that does not correspond to the real name and have it displayed next to the MAC ADDRESS in the devices?
It would be convenient to be able to immediately identify the NUT without having to remember the MAC, especially for new devices.

For example:
CXXXXXXXXXX0 with real name nut> friendly name Rocco
FXXXXXXXXXX3 with real name nut> friendly name Moana
DXXXXXXXXXX3 with real name nut> friendly name Selen

It's possible?

Thanks in advance

shogunxam, 2021-03-20:
Hi @SmartM-ui,
possible is possible I have to study how to structure the interface.

SmartM-ui, 2021-03-20:
Thanks for your work, I'm sure you will succeed!

PR #41 (2021-03-23): Add support for device friendly name
Close #32
closed 2021-03-24 by shogunxam