shogunxam/ESP32_BLETracker #34: Retain State and Battery level after black-out

SmartM-ui, 2021-03-19:
Hi,
would it be possible to implement a retain on the home / not home state and on the last battery level?

If I'm not mistaken, after a blackout, if the NUT is out of the house, the values are "unknown"

Thanks

shogunxam, 2021-03-20:
I'm not quite sure enabling the retain on the state of the device is a good idea. In example take a look to this sequence:
1. your BLE device is at home and discovered
2. there is a power outage
3. you and your BLE Device go out of home
4. the power is restored (you are still out of home)
In this scenario if the retain is enabled  your broker will tell to all the subscribers your device is at home.
What I can do is, only when the white list is enabled, initializing the list of discovered devices with all the devices in the white list, with initial state off so that all the devices not present when the power is restored will be considered off and not "not available".

SmartM-ui, 2021-03-20:
Hi,
it seems to me a great idea!

PR #39 (2021-03-20): Set devices state to off at startup
Close #34
closed 2021-03-24 by shogunxam