shogunxam/ESP32_BLETracker #46: Last battery level status

SmartM-ui, 2021-03-31:
Hi Mario,
it is possible to memorize the last battery status of the NUT, detected by the ESP, so as to make it available even if the ESP should restart (with the device tracker not home)?

Thanks

shogunxam, 2021-04-10:
Hi, sorry but I don't see the usefulness of this feature. If the device is away from home, an untrue data would be reported; from the panel you could read that the device is charged, when instead it could be completely discharged. In this case, in my opinion, it is more appropriate to report an unknown status (-1), which happens even if the ESP does not reset and the device is not found when the battery reading timeout is triggered.

SmartM-ui, 2021-04-10:
Hi,
having found myself several times in the situation of being away from home with the ESP restarted, I can confirm that having the last data on the battery, now that the date and time of the last detection is also displayed, is useful.

Could you find a meeting point and display the last battery level (with the detection date) with the (-1) next to it so you understand that it is not certain?

Thanks again for the great job done

shogunxam, 2021-04-11:
If you are using Home Assistant you can achieve the same result using an input_number and an automation i.e.

```
sensor
- platform: mqtt
  state_topic: 'home/BLETracker/FFEEDDCCBBAA'
  name: 'nut_battery' 
  value_template: "{{ value_json.battery|int }}"
  force_update: true

input_number:
 battery_value:
     name: BatteryValue
     min: -1
     max: 100
     unit_of_measurement: "%"

automation
- id: update_nut_battery
  alias: update_nut_battery
  initial_state: true
  trigger:
    platform: state
    entity_id: sensor.nut_battery
  condition:
    condition: numeric_state
    entity_id: sensor.nut_battery
    above: -1
  action:
    service: input_number.set_value
    target:
      entity_id: input_number.battery_value
    data:
      value: "{{ states('sensor.nut_battery')}}"
```

Hope this can help
Best regards

SmartM-ui, 2021-04-14:
Hi Mario,
having the data in MQTT, would it be possible to pass to HA also the date and time of the last battery detection, without having to change the firmware on the ESP?

Thanks in advance

shogunxam, 2021-04-17:
Hi,
 you can use a rest sensor to retrieve the information you need. Unfortunately there is a bug in the current released version preventing the rest sensor to work, but I have fixed it in the develop branch ( PR #48 )
 ```
 sensor:
  - platform: rest
    resource: http://<bletracker_ip>/getsysinfodata
    name: LastBatteryReading
    value_template: >-
      {% if value_json.devices | length > 0 %}
        {% for x in value_json.devices %}
          {% if x.mac ==  "AABBCCDDEEFF"%}
            {{x.bttime}}
          {%endif%}
        {% endfor %}
      {% endif %}
 ```
You should improve the template to check for missing bttime field in order to avoid error.

SmartM-ui, 2021-04-17:
Tank you BIG Mario!

SmartM-ui, 2021-04-18:
> Hi,
> you can use a rest sensor to retrieve the information you need. Unfortunately there is a bug in the current released version preventing the rest sensor to work, but I have fixed it in the [...]

Thanks again Mario for the work done.
Place the perfectly working code:

sensor:
- platform: rest
  resource: http://<bletracker_ip>/getsysinfodata
  name: LastDateBatteryReading
  username: "username"
  password: "password"
  authentication: basic
  headers:
    User-Agent: Home Assistant
    Content-Type: application/json
  value_template: >-
    {% if value_json.devices | length > 0 %}
      {% for x in value_json.devices %}
        {% if x.mac ==  "AABBCCDDEEFF"%}
          {% if x.bttime is defined %}
            {{x.bttime | timestamp_local}}
              {%else%}
                0
          {%endif%}
        {% endif %}
      {% endfor %}
    {%endif%}
  device_class: timestamp

input_number:
 battery_number:
     name: battery_inputnumber
     min: -1
     max: 100
     unit_of_measurement: "%"

input_datetime:
  battery_datetime:
    name: battery_datetime
    has_date: true
    has_time: true

automation:
- id: update_nut_battery
  alias: update_nut_battery
  trigger:
    platform: state
    entity_id: sensor.lastdatebatteryreading
  condition:
    condition: numeric_state
    entity_id: sensor.nut_battery
    above: -1
  action:
    service: input_number.set_value
    target:
      entity_id: input_number.battery_number
    data:
      value: "{{ states('sensor.nut_battery')}}"

- id: update_datetime_nut_battery
  alias: update_datetime_nut_battery
  trigger:
    platform: state
    entity_id: sensor.lastdatebatteryreading
  condition:
    condition: template
    value_template: "{{as_timestamp(states('sensor.lastdatebatteryreading')) > 0}}"
  action:
    service: input_datetime.set_datetime
    target:
      entity_id: input_datetime.battery_datetime
    data:
      timestamp: "{{as_timestamp(states('sensor.lastdatebatteryreading'))}}"

shogunxam, 2021-04-25:
I suppose I can close this issue.

closed 2021-04-25 by shogunxam