shogunxam/ESP32_BLETracker #51: New firmware release notification

SmartM-ui, 2021-04-25:
Hi Mario,
from the ESP web panel it appears when you release a firmware update.
Could you implement the ability to receive an event notification?
Thank you

shogunxam, 2021-05-01:
Hi @SmartM-ui sorry if i reply late but I've been busy with my job.
All the information you need are already available. The rest API  GET /getindexdata returns a JSON with the current firmware version in the "ver" property and the GitHub API  GET [https://api.github.com/repos/shogunxam/ESP32_BLETracker/releases/latest](https://api.github.com/repos/shogunxam/ESP32_BLETracker/releases/latest) returns a JOSN with the information related to the latest release, here you can look at to  the "tag_name" property.
In HA  you can use two REST sensors to retrieve those information and compare them in order to have an alert.
```
sensor:
 - platform: rest
   resource: http://<bletracker_ip>/getindexdata
   name: myFirmwareVersion
   scan_interval: 86400
   value_template: >-
     {{value_json.ver|trim("RrVv ")|float}}
 - platform: rest
   resource: https://api.github.com/repos/shogunxam/ESP32_BLETracker/releases/latest
   name: LatestFirmwareVersion
   scan_interval: 86400
   value_template: >-
     {{value_json.tag_name|trim("RrVv ")|float}}

automation:
- id: notifyNewFirmawareVersion
  alias: notify new firmaware version
  trigger:
    platform: state
    entity_id: sensor.LatestFirmwareVersion
    condition: template
    value_template: "{{states('sensor.LatestFirmwareVersion')) > states('sensor.myFirmwareVersion')) }}"
action:
   .....
```

I haven't tried it but i think it should work. 
Hope this can help you, cheers.

SmartM-ui, 2021-05-01:
Hi Mario, THANK YOU!
I made a few small changes to the yaml.
I report my configuration, if it should be useful to others:

'''
sensor:
- platform: rest
  resource: http://<bletracker_ip>/getindexdata
  name: ESP32_BLEtracker_Firmware_Version
  username: "myuser"
  password: "mypass"
  scan_interval: 86400
  authentication: basic
  headers:
    User-Agent: Home Assistant
    Content-Type: application/json
  value_template: >-
    {{value_json.ver|trim("RrVv ")|float}}
- platform: rest
  resource: https://api.github.com/repos/shogunxam/ESP32_BLETracker/releases/latest
  name: ESP32_BLEtracker_Last_Firmware_Version
  scan_interval: 86400
  value_template: >-
    {{value_json.tag_name|trim("RrVv ")|float}}

automation:
- id: esp32_bletracker_new_firmware_version
  alias: ESP32_BLEtracker_NEW_Firmware_Version
  trigger:
    platform: state
    entity_id: sensor.esp32_bletracker_last_firmware_version
  condition:
    condition: template
    value_template: "{{states('sensor.esp32_bletracker_last_firmware_version') > states('sensor.esp32_bletracker_firmware_version') }}"
  action:
    service: notify.telegram
    data_template:
      title: ESP32 BLEtracker NEW Firmware
      message: "New version available {{ states('sensor.esp32_bletracker_last_firmware_version') }}"
'''

closed 2021-05-01 by SmartM-ui