shogunxam/ESP32_BLETracker #52: Multiple esp32 network

tux240, 2021-04-28:
Hi,

Thanks for this project ! 
Is actually possible to connect multiple ESP32 like a mesh network ?
Like PainlessMesh (https://gitlab.com/painlessMesh/painlessMesh) ? 
Thanks

shogunxam, 2021-04-29:
Hi tux240,
currently you can have multiple esp32 with different gateway name but they are not connected between them, each of them doesn't know the others and all of them talk with the broker.
If you think this can be useful I can try to add this feature if it's not too much memory consuming.

tux240, 2021-04-29:
Thanks for your answer.
Ok, I understand. I think it could be useful to track Beacons over big houses or another location without adding wifi repeaters.
Nevertheless, I don't know if it can consume too much memory... I have installed Painless library on 5 ESP32 and works like a charm but doesn't yes test with more functionalities...

I have seen some others projects with trilateration to get the precise beacon location.
Do you think this project could to do it ? (like https://github.com/simonbogh/ESP32-iBeacon-indoor-positioning)
My scenario is to track shoes boxes on the house :p

shogunxam, 2021-05-01:
Hi @matthieup240 sorry for the delay, I was busy with my job.
I don't think my project can work  with [ESP32-iBeacon-indoor-positioning](https://github.com/simonbogh/ESP32-iBeacon-indoor-positioning) as it is simply because the payloads format I send via MQTT are different, but looking to that project I'm sending all the information you need (MAC address and rssi) so theoretically it could work with some changes to the  ESP32-iBeacon-indoor-positioning dashboard to get the needed information also from my payloads.
Let me do a deeper analysis to integrate the Painless library in my project.

closed 2021-07-07 by stale[bot]