shogunxam/ESP32_BLETracker #57: Implement a Mailbox Sensor,Led and MQTT message

rcarreira88, 2022-06-19:
Hello, can you help me implement this with your code?
Thank you

shogunxam, 2022-06-20:
Hi @rcarreira88, could you please explain a bit more your request? Thx

rcarreira88, 2022-06-21:
Hi, @shogunxam,
i want to put your code in my mailbox esp32 to send a MQTT command to open door if it detects my smartband MAC address ,
but i also want:
Switch1 connects to GPIO and sends MQTT message and turns on LED connected to a GPIO,
Switch2 connects to GPIO and sends MQTT message and turns off LED.

shogunxam, 2022-06-22:
If I have understood you want integrate my code in an existing project. I hope you are a software developer because the work could not so easy if you haven't the proper knowledge.
I think you have to give up the idea to have a web server because my code has already reached the limit of the ESP32 memory.
The project uses this external libraries:
- PubSubClient@2.8
- Regexp@0.1.0
- For the BLE library I used this one  https://github.com/espressif/arduino-esp32/tree/6b0114366baf986c155e8173ab7c22bc0c5fcedc/libraries/BLE. I think you can use also the latest
released with the ArduinoESP32 platform. If you are using platformIO I suggest to you to copy from my platformio.ini

I tried to make the project modular, but the code become a little messed up after the introduction of the support for the FHEM platform.
Main modules are:
 - WiFiManager -> used to access the WiFi
 - mqtt_client -> Used to connect to the MQTT broker and publish messages
- There are other utility modules such as mutex or NTPTime to get the time and date from a server.
- In the main module there is the code to scan the BlueToothLE, the core is composed by two pieces one is the class MyAdvertisedDeviceCallbacks where each advertised device is processed, then in the loop function there is the code executing the scan (search for the macro PROGRESSIVE_SCAN). I made two kind of scan one progressive with limited time search and one searching for all the available devices in a longer time range.

If you need also the battery status of the devices follow the batteryTask function.
If you need something simpler take a look to this project
https://github.com/sdesalve/dss_mqtt_binary_sensor_ble_scanner
I started from this project and I evolved it.

About the GPIO I have never worked with them, but googling, I'm sure you will find what you need, in example here you can find how to turn on a led.
https://microcontrollerslab.com/gpio-pins-esp32-led-blinking-example/

closed 2022-09-16 by stale[bot]