shogunxam/ESP32_BLETracker #59: reset

leonel85, 2023-09-19:
Hello. I try to compile in debug mode
the only thing that I change in platformio.ini is:

[com]
esp32_platform = **espressif32**
#arduino-esp32 1.0.5-rc7
#esp32_platform = https://github.com/shogunxam/platform-espressif32.git#13a6a65fda58242ba3e10a7663fe07a2163c9f2f

My ESP32 just reebot/start and reboot

I have no messages on serial I try to print a string on 
void setup() ... but it just reboot without messages

Decoder output is :

0x4009a1a9: abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/abort.c:38
0x401e5337: rcUpdatePhyMode at ??:?
0x401e537e: rcUpdatePhyMode at ??:?
0x401e5477: rcUpdatePhyMode at ??:?
0x401e5cf5: rcUpdateRate at ??:?
0x400db863: MyAdvertisedDeviceCallbacks::onResult(BLEAdvertisedDevice) at D:/Develop/Software/Local/ESP32_BLETracker-master/main/main.ino:152   
0x400dbab9: _GLOBAL__sub_D_trackedDevicesMutex at D:/Develop/Software/Local/ESP32_BLETracker-master/main/main.ino:589
0x400dbb4b: _publishToMQTT(char const*, char const*, bool) at D:\Develop\Software\Local\ESP32_BLETracker-master/main/mqtt_client.cpp:43 (discriminator 8)
0x400dc60c: StringSumHelper::StringSumHelper(char const*) at D:\Develop\Software\Local\ESP32_BLETracker-master/main/settings.cpp:147
  \-> inlined by: Settings::toJSON() at D:\Develop\Software\Local\ESP32_BLETracker-master/main/settings.cpp:163
0x400f65c3: String::changeBuffer(unsigned int) at C:/Users/manjate/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.cpp:668 (discriminator 3)

Thanks in advanced

shogunxam, 2023-09-20:
Hi leonel85.
When I started to develop this project I found some issues in the espressif32 platform coming with platformIO related to the BLE support, for this reason I forked the espressif32 branch and applied some patch.
Time is passed and the things could be changed and my code could not be compatible with the version you are using.
Do you really need to change the platform?
Debuging it will require a lot of time. From a first look it seems something related to the string allocation.

leonel85, 2023-09-21:
Hi
Thanks
I have develop another things that I have to mix together. If it’s possible I’d like to use official framework because I have already use it and it could be a little bit better for future releases 

I don’t real need to debug everything. I think that this is a good project for me and maybe it already works with the new platform
It could be a good help understand how to debug and fix it

shogunxam, 2023-09-22:
As I have already said seems to be related to the string reallocation, perhaps the new esp32 library has introduced a memory leak causing the incapacity to reallocate the space for new string.
As first step you could try to reserve enough space for the String data located in the function Settings::toJson. Otherwise you have to change the code making it a char array and change the method used to concatenate the various part of the string.
Let me know if this can solve. Meanwhile this weekend, if I found time, I'll try to take a look to it.

leonel85, 2023-09-22:
Hi 
I see that that function is used by OTA so first I disable OTA but I had another error
I debug and find that the problem was the White list
BLE_KNOWN_DEVICES_LIST
I just remove it from initialisation 
Now it works but after a while it crashes 
It find a lot of devices
I have to debug a little bit 
Sorry I’m writing from mobile. I will give you more details later
Thanks for help

shogunxam, 2023-09-22:
Are you sure you have changed only the platform?
Changing only the platform the build fails

shogunxam, 2023-09-22:
Looking to the memory used by the toolchain  I suspect the problem is there:
With the original tool chain we have
RAM:   [==        ]  18.0% (used 59064 bytes from 327680 bytes)
Flash: [========  ]  78.6% (used 1545945 bytes from 1966080 bytes)

with the latest expressif32
RAM:   [==        ]  20.1% (used 65860 bytes from 327680 bytes)
Flash: [========  ]  84.7% (used 1665821 bytes from 1966080 bytes)

I fear there is no more enough space for the dynamic allocation

shogunxam, 2023-09-23:
I I'm amazed at how it's worked so far.
I forgot the return in the assignment function
`Settings::KnownDevice &Settings::KnownDevice::operator=(const KnownDevice &dev)
{
    readBattery = dev.readBattery;
    memcpy(address, dev.address, ADDRESS_STRING_SIZE);
    memcpy(description, dev.description, DESCRIPTION_STRING_SIZE);
    return *this;
}`

Adding it has resolved the crash

closed 2023-09-23 by shogunxam