shogunxam/ESP32_BLETracker #62: Feature Request: Start scan via html request

hismastersvoice, 2024-05-19:
Hello shogunxam,

I use your software now for some month an it works great. Thanks for that at first.

In my hole house I have installed 4x ESP32 with the software, and the detection works great.
For an perfect automation it would be helpful to start the scan via command like http://ip/scan or equal.
Use case is e.g. with opening of the front door or garage, I'd like to start directly the scan to know the user at the door. 

I'm not a good programmer in C, so would it be able that you can help me?

Thanks in advance.

HisMastersVoice

shogunxam, 2024-05-20:
Hi @hismastersvoice, I am glad you appreciate my work.
I don't see any technical problems implementing your request, other than the code would get a little bigger, but I honestly don't see the benefit.
Could you explain in more detail why a continuous scan isn't right for you?
This would help me understand if this is a change that other users might need as well.

hismastersvoice, 2024-05-20:
Good moring @shogunxam,

thanks for you fast response.

The continuous scan is good as it is.

My need is...
- I start different action by entering the house or garage, defined to the right person.
- For battery saving reason, I don't what to do scanning <10 sec.
- If I open the door, in worst case it takes 10 sec. before the action is starting.
- For me this is sometimes to long

Now I want to start scanning by opening the door contact or when NFC access is granted.
Then I got in 1-2 sec. the right person, and can start action(s).

At the moment I only can do not personalized actions.

Thanks.

shogunxam, 2024-05-20:
I have created the branch Manual_start_stop_scan. I haven't tested it yet, just wrote  a first draft of the code.
I think I'll have time to work on it this weekend.
If you want give it  a try, change the line 106 of the settings.cpp
from 
`manualScan = eManualSCanMode::eManualSCanModeDisabled;`
to
`manualScan = eManualSCanMode::eManualSCanModeOff `

Than you should be able to start and stop the scan using the api
```
GET  http://<device_ip>/scan?on=1
GET  http://<device_ip>/scan?off=1
```
I think also 
```
GET  http://<device_ip>/scan?on
GET  http://<device_ip>/scan?off
```

should be enough.

But I don't guaratee it works :smiley:

hismastersvoice, 2024-05-20:
Thanks !
Will try it next days, and come back.

hismastersvoice, 2024-05-20:
on=1 turned on the scan
off=1 turn off, but than starts spamming MQTT with a lot of messages. 

Will have a closer look this days.

shogunxam, 2024-05-22:
I have updated the branch adding a new check box in the configuration page to enable/disable the manual scan (by default is disabled) and than when the scan is off the device state is published only every two minutes.
Could you please give a try and let me know what do you think?

hismastersvoice, 2024-05-22:
Thanks for changing...
I'm on a business trip until Friday morning, test on Friday afternoon.

hismastersvoice, 2024-05-24:
@shogunxam 
Thanks for working on a solution! 

Works like you describe...
But the result of the scan is also submitted after e.g. 10 sec., so if I scan automatically or  manual is the same timing.
Or do I understand the function wrong?

shogunxam, 2024-05-24:
When the scan is enabled in auto or manual mode the timing of the publication is the same.
When the scan if manual but off, to avoid to spam on MQTT (because there is no more the delay of the scan) the publication of the state is sent every 2 minutes. Obviously when scan is off the BLE device will be considered non present.

hismastersvoice, 2024-05-24:
"If this is sufficient for your needs I'll merge the changes in the master branch."
Yes an no... ;)

My idea was to get faster Information about presence, now I had in both situations the 10 sec. or however the scan time is set.
When I look with my little knowledge to the code, the scan result are submitted after the full scan-period e.g. 10 sec., and not after new device was found. I'm not sure, but it looks like that I only can set down the scantime, and start / stop by a logic to get faster results.

Is it possible the the manual scan stops after every loop automatically?. 
So I don't have to set two commands.
Also it is more reliable if the scan?off=1 not reach the controller.

hismastersvoice, 2024-05-24:
Thanks for explanation...
I habe installed today an additional motion detector, so the scan starts some second before the user reach the door.
So together with the door contact it fits now perfect with the manual scan an the 10 sec scan time.

Thanks again for do the changes special for me. 
Now I have the perfect solution!!!!!!!  ;)

hismastersvoice, 2024-05-24:
Also I have change the Hardware today to XIAO ESP32-C3 with external Antenna and the scan radius is amazing.
I only need only 2 instead of 4 devices to cover the house and garage.

closed 2024-10-27 by shogunxam