shogunxam/ESP32_BLETracker #67: WLAN / IP changeable in WebUI?

hismastersvoice, 2024-08-31:
Hello,

I'm absolutly happy with your BLE Tracker. Works great.
I also like to use it for my Friends homes, but there is a small issue for them/me.

The SIDD/Passsword and IP is only to set by Code and not by WebUI.
Is there a possibility to set it by WebUI.
Like Tasmota to open a AP to the WLAN and IP?

If you can do it would be a big step, I can than use an ready compiled Image files to flash it on the ESP and setup with a phone...

Is it a big Thing?
Thanks in advance.

shogunxam, 2024-09-01:
Hi, I have begun the development of that feature some time ago but then, because some problem and the limited time to work on it, I stopped it.
As soon as I'll have time I'll restart the work on it.

Thank you very much to appreciate my work.

hismastersvoice, 2024-09-01:
Big thanks for your help!

closed 2024-10-27 by shogunxam