shogunxam/ESP32_BLETracker #71: Trouble building fhem version

Jo23674, 2025-02-27:
Thanks a lot for this FW. That's exactly what I was looking for - especially the fhem connection via lepresenced. 
I successfully built the MQTT version using PlatformIO (on Linux) and it works flawlessly (even though I'm a noob). My devices are discovered and I do receive the mqtt messages in fhem.
However, when I try to compile either the fhem version or the fhem debug version, I end up in several error messages which I do not really understand.

`/home/johannes/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev-ble-fhem-debug/libFrameworkArduino.a(main.cpp.o):(.literal._Z8loopTaskPv+0x8): undefined reference to `setup()'
/home/johannes/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev-ble-fhem-debug/libFrameworkArduino.a(main.cpp.o):(.literal._Z8loopTaskPv+0xc): undefined reference to `loop()'
/home/johannes/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev-ble-fhem-debug/libFrameworkArduino.a(main.cpp.o): in function `loopTask(void*)':
/home/johannes/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42: undefined reference to `setup()'
/home/johannes/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/johannes/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:48: undefined reference to `loop()'
collect2: error: ld returned 1 exit status
*** [.pio/build/esp32dev-ble-fhem-debug/firmware.elf] Error 1`

Is there anything I forgot when compiling or any other idea what could be wrong?

shogunxam, 2025-02-28:
Hi, thanx to use my firmware.
If I have done all the things right I should have built the fhem version too.
Anyway tomorrow morning I'll check it.
Please take a look if, by mistake, you have changed the code. It looks like there is some curly bracket placed in the wrong place or something like that.

shogunxam, 2025-03-01:
I tried compiling for fhem using the latest firmware version and it works like a charm.
I'm using PlatformIO Core 6.1.17·Home 3.4.4

Jo23674, 2025-03-01:
Thanks a lot for the quick support! As I did not change anything except the user config, I assume it's my environment which might cause the issue. I'll have a look into it, try from scratch and let you know. I'm also using PlatformIO Core 6.1.17, Home 3.4.4 (on an ubuntu machine).

Jo23674, 2025-03-01:
Quick update: I finally managed to compile without errors. If I start from scratch with a fresh cloned project and only edit this one line in the .ini file to 
default_envs = esp32dev-ble-fhem-release
it works. Changing it afterwards from within platformIO does not work. 
Again, thanks a lot!

closed 2025-03-01 by Jo23674