shogunxam/ESP32_BLETracker #72: Crash on ESP32

jimjaeger, 2025-04-17:
The ESP32_BLE tracker crashes in a loop after about a minute.

I configured the following:
esp32_platform = espressif32
board = esp32dev
- SSID / WifiPassword
- MQTT Broker
- BLE Whitelist (but also without configuring a list, it crashes)
and build and upload it with PlatformIO to a ESP32.
The initial build run in error so I had to add build_flag -fexceptions to get it compiled successfully.
After upload I rebooted the ESP32 and the webserver became available. 2 devices have been recognized but only the general MQTT message arrive in MQTT server. After a short time the ESP crashes and reboots again and again.

To recreate I recompiled and upload the debug version and followed the steps to debugging.

1970-01-01 00:00:05 loaded TZ UTC
1970-01-01 00:00:10 SPIFFSLogger  first_record 0, num_written_record 107,  maxLogsNum 200
1970-01-01 00:00:15 Connecting to WiFi EF16510...1970-01-01 00:00:20 
1970-01-01 00:00:25 TZ is UTC
1970-01-01 00:00:30 --------------------
1970-01-01 00:00:35 Connected to <removed>
1970-01-01 00:00:40 IP address: <removed>
1970-01-01 00:00:50 Reset due to power-on event
1970-01-01 00:01:00 WebServer Setup
1970-01-01 00:01:05 mDNS responder started
1970-01-01 00:01:11 INFO: Connecting to MQTT broker: <removed>
1970-01-01 00:01:16 {"version":7,"mqtt_enabled":true,"wifi_ssid":"<removed>","wifi_pwd":"<removed>","wbs_user":"admin","wbs_pwd":"admin","tz":"UTC","gateway":"BLETracker","mqtt_address":"<removed>","mqtt_port":1883,"mqtt_usr":"","mqtt_pwd":"","scanPeriod":10,"maxNotAdvPeriod":120,"loglevel":2,"manualscan":false,"whiteList":true,"trk_list":{"<removed>":{"battery":true,"desc":"Orange"},"<removed>":{"battery":true,"desc":"Tag_rt"}}}
1970-01-01 00:01:21 Error: MQTT broker disconnected, connecting...
1970-01-01 00:01:26 INFO: The client is successfully connected to the MQTT broker
1970-01-01 00:01:36 INFO: MQTT message published successfully, topic: home/BLETracker/LWT , payload: online , retain: True 
INFO: Running mainloop
1970-01-01 00:01:46 Main loop Free heap: 75852
1970-01-01 00:01:51 Number device discovered: 2
1970-01-01 00:01:58 INFO: Tracked device discovered again, Address: <removed>, RSSI: -50
1970-01-01 00:02:06 INFO: MQTT availability topic: home/BLETracker/LWT
1970-01-01 00:02:11 INFO: MQTT message published successfully, topic: home/BLETracker/LWT , payload: online , retain: True 
1970-01-01 00:02:16 INFO: MQTT availability topic: home/BLETracker/LWT
1970-01-01 00:02:21 ERROR: MQTT message not published, either connection lost, or message too large. Topic: home/BLETracker/LWT , payload: online , retain: True 
1970-01-01 00:02:37 INFO: MQTT availability topic: home/BLETracker/LWT
1970-01-01 00:02:42 INFO: Connecting to MQTT broker: <removed>
1970-01-01 00:02:52 Error: MQTT broker disconnected, connecting...
1970-01-01 00:02:57 INFO: The client is successfully connected to the MQTT broker
1970-01-01 00:03:07 INFO: MQTT message published successfully, topic: home/BLETracker/LWT , payload: online , retain: True 
1970-01-01 00:03:12 INFO: MQTT message published successfully, topic: home/BLETracker/LWT , payload: online , retain: True 
1970-01-01 00:03:22 
Reading Battery level for <removed>: Retries: 0
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (hciT)
Core  0 register dump:
PC      : 0x40096ca7  PS      : 0x00060036  A0      : 0x40095b13  A1      : 0x3ffe8a50  
A2      : 0x3ffc5f44  A3      : 0x3ffc5f80  A4      : 0x00000001  A5      : 0x00000000
A6      : 0x3ffc5f80  A7      : 0x00000016  A8      : 0x3ffc5f4c  A9      : 0x80000001
A10     : 0x3ffe7c2c  A11     : 0x3ffe7c2c  A12     : 0x00000014  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x80000001  SAR     : 0x00000015  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x400918e0  LEND    : 0x400918f6  LCOUNT  : 0xffffffff

Backtrace: 0x40096ca4:0x3ffe8a50 0x40095b10:0x3ffe8a80 0x40095ac2:0xa5a5a5a5 |<-CORRUPTED

ELF file SHA256: 0000000000000000

I executed the debug script and get the following stack:

stack:
0x40096ca4: taskSelectHighestPriorityTaskSMP at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3435
  \-> inlined by: vTaskSwitchContext at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3519        
0x40095b10: _frxt_dispatch at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port/xtensa/portasm.S:436     
0x40095ac2: _frxt_int_exit at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port/xtensa/portasm.S:231

shogunxam, 2025-04-18:
> 1970-01-01 00:01:16 {"version":7,"mqtt_enabled":true,"wifi_ssid":"","wifi_pwd":"","wbs_user":"admin","wbs_pwd":"admin","tz":"UTC","gateway":"BLETracker","mqtt_address":"","mqtt_port":1883,"mqtt_usr":"","mqtt_pwd":"","scanPeriod":10,"maxNotAdvPeriod":120,"loglevel":2,"manualscan":false,"whiteList":true,"trk_list":{"":{"battery":true,"desc":"Orange"},"":{"battery":true,"desc":"Tag_rt"}}}
> 1970-01-01 00:01:58 INFO: Tracked device discovered again, Address: , RSSI: -50
> Reading Battery level for : Retries: 0

Hi jimjaeger, the crash seems to be in the esp32 library. Are you using the original configuration or have you updated it to the latest esp32 library?

jimjaeger, 2025-04-18:
Hi, thanks for the fast response.

I resetted the platformio setup and verified that the correct board is selected and now it works stable.

I can confirm that the tracker works with Gigaset G-tag Beacon (including battery information).

closed 2025-04-18 by jimjaeger

jimjaeger, 2025-04-18:
Thanks for integrating homeassistant mqtt publish option and examples. Do you already think about sending homeassistant discovery information too so that homeassistant automatically integrate the it as device?

shogunxam, 2025-04-18:
Glad to hear you've solved it. At the moment, automatic integration with Home Assistant is out of scope, but it could be a nice-to-have for one of the next releases. I just need to find the time to work on it!