shogunxam/ESP32_BLETracker #75: Access Point question

hismastersvoice, 2025-04-22:
Good Morning,

I have one question to the Access Point Mode.

-> No WLAN = Access Point is starting. Right?
Is there a timeout that the Access Point Mode is restart the ESP after e.g. 15 Min.

My question belongs to the follow case that happen to me. 
My Router and Access Points in the house got a new firmware, so the WLAN stops for some minutes.

After everything was ready, both of my tracker where in AP Mode.
After a power reset they connect to back to the WLAN.

Is this normal, or only an issue with my system?

shogunxam, 2025-04-23:
> Thanks in advance...

Hi, it's a bug. I have discovered it me too just this week. It should be solved in the develop branch. When the esp loose the connection with the Wifi and fails to reconnect it enters in AP mode and it remains in that mode until a restart. The fix consists in retring the connection to the WiFi after 1 minute if a non empty SSID is configured. If it fails it stays in AP mode and it retry after a minute and so on.

I'm doing some test also for other new features I hope to make a new release soon.

shogunxam, 2025-04-24:
Fixed with the commit 1d9cc363b54e2703a489808c363e66331faa3d98

closed 2025-04-24 by shogunxam