shogunxam/ESP32_BLETracker #77: UDP Message instead of MQTT

hismastersvoice, 2025-05-03:
Hello shogunxam

I have played around with your code to get the messages via UDP.

Why... 
In my project I offen use Loxone MiniServer, this device has no MQTT Broker, so I have to use additional hardware.
If I get the messages direct via UDP, I can us the data without any additional hardware.
 
Message each loop 
`module=presence.west XXXXXXXXXXXXXX=1 rssi=-70 battery=100 timestamp=1746283293`
`module=presence.west XXXXXXXXXXXXXX=0 rssi=-68 battery=100 timestamp=1746281586`

Message every 120 Sec. (SYS_INFORMATION_DELAY 120)	
`module=presence.ost alive uptime=1.19:21:21 SSID=HMV  rssi=-69 IP=192.168.136.120 MAC=XX:XX:XX:XX:XX:XX timestamp=1746283405`

It works fine for me...
But don't what to use a unprofessionell code /I'm a bad coder ;)/
Do you see any chance to put UDP in to your code?

I only have put in main.ino, but I think it is better to have it separated like MQTT/FHEM...

```
void sendBLEStateUDP(const char address[ADDRESS_STRING_SIZE], const char state[4], int8_t rssi, int8_t batteryLevel)
{
  unsigned long now = NTPTime::getTimeStamp();
  String wifi_mac = WiFi.macAddress();
  wifi_mac.replace(":", "");
  udp.beginPacket(SettingsMngr.udpServer.c_str(), SettingsMngr.udpPort);
  udp.print(String("module=") + SettingsMngr.location + String(".") + SettingsMngr.gateway + String(" ") + address + String("=") + state + String(" rssi=") + rssi + String(" battery=") + batteryLevel + String(" timestamp=") + now);
  udp.endPacket();
}

void sendSySInfoUDP()
{
  unsigned long now = NTPTime::getTimeStamp();
  long rssi = WiFi.RSSI();
  static String IP;
  IP.reserve(16);
  IP = WiFi.localIP().toString();
  char strmilli[20];
  String wifi_mac = WiFi.macAddress();
  udp.beginPacket(SettingsMngr.udpServer.c_str(), SettingsMngr.udpPort);
  udp.print(String("module=") + SettingsMngr.location + String(".") + SettingsMngr.gateway + String(" alive uptime=") + formatMillis(millis(), strmilli)  + String(" SSID=") + WIFI_SSID + String(" rssi=") + rssi + String(" IP=") + IP.c_str() + String(" MAC=") + wifi_mac + String(" timestamp=") + now);
  udp.endPacket();
} 
```

shogunxam, 2025-05-03:
Hi I'll try to take a look in to it as soon as I have time 
I suppose the UDP object is an instance of the WiFiUDP class right?

hismastersvoice, 2025-05-03:
Thanks for your help... 

Yes, WiFiUDP class

#include "WiFiUdp.h"
WiFiUDP udp;

Setup
udp.begin(UDPPORT)

PS: The new WebUI in Brach Development look great!!

shogunxam, 2025-05-03:
About the messages sent, are there some restrictions?
I see in the first you don't send the last battery reading time, perhaps I can add it.
In the second message I see you send the ip of the esp32. Do you need it? The receiver should already know who is sending the data.
Are 120 seconds mandatory? In the last version I'm sending the MQTT availability topic every 300 sec.
Is the timestamp required in both the messages?

The challenging part will be the UI to configure the UDP protocol. My rule is: don't send data you don't need, so the interface sould become dynamic based on the compilation parameters, and I don't want keep a duplicate of the common part. So I have to find a way to don't send the MQTT tab when is not used and the same for the UDP configuration tab.
In the UDP configuration tab I'll put the remote server IP and remote port.
I don't think there is reason to keep both MQTT and UDP enabled I think they are mutually exclusive.

hismastersvoice, 2025-05-03:
No there are no restrictions.

Last battery read could be helpfull.
The 120 sec. are use becouse it was set als value in my used code. If it is 300 sec it is also OK.

For me is ok only to send UDP or MQTT. 
I think there is no need to send both.

hismastersvoice, 2025-05-03:
Timestamp is needed in alive message to check ever x minutes if the ESP32 is running.

hismastersvoice, 2025-05-04:
Maybe that is an Option...
Compile MQTT and UDP, and let the user choose in the WebUI.

https://github.com/user-attachments/assets/99678a02-6d07-415f-b925-8ea193ed13e2

shogunxam, 2025-05-04:
I have merged the changes in develop. Give it a try.
You can enable the UDP it in the config.h or building  the proper variant in PlatformIO `esp32dev-ble-udp-*`

hismastersvoice, 2025-05-04:
Short testing...  Works fine
Thanks for the implementation.

closed 2025-05-18 by shogunxam