shogunxam/ESP32_BLETracker #79: Branch: develop - Saving Config is extremly slow

hismastersvoice, 2025-05-07:
Hello shogunxam,

I play around with the UDP function in the dev branch.
What I have notice is that for the first run the saving of the config is extremly slow.
Every value takes exactly 5 sec. to save. So maybe this is a loop timer that makes it so slow.
Not sure if this is also at the master, but here I have notice it.

`
1970-01-01 00:02:**50** ssid = XXXXXXXXXXX

1970-01-01 00:02:**55** wifipwd = XXXXXXXXXXXX
1970-01-01 00:03:**00** gateway = ESP32-XXXXXXXXXXX
1970-01-01 00:03:**05** wbsusr = admin 
1970-01-01 00:03:**10** wbspwd = admin 
1970-01-01 00:03:**15** mqttsrvr =  
1970-01-01 00:03:**20** mqttusr =  
`

shogunxam, 2025-05-08:
In AP mode the main loop run faster than in WiFi mode because it do nothing. Because you are running in debug mode the messages are sent to the serial. I suppose the delay should depend by the thread used by the serial. Perhaps it has lower priority than the main thread and this one takes the CPU for the most of the time making the print slower. You shouldn't have tis problem in release. At the moment I don't find other explanations.

hismastersvoice, 2025-05-08:
Your right... It happens only with Serial turned on.
So everything is fine ;)

closed 2025-05-18 by shogunxam